prusa3d/Prusa-Firmware-SL1 #1: Source for octo-uploadd.py

njcrawford, 2019-10-03:
I'm interested in adding partial support for Octoprint's printer and job APIs. As best I can tell, the Octoprint-like interface is provided by octo-uploadd.py, but I can't seem to find it anywhere on Github. (Even Google wasn't able to help!)

Could someone kindly point me in the right direction?

vladamatena, 2019-10-03:
The Prusa-Firmware-SL1 repository contains recipes used to build a custom Linux distribution based on the Yocto project. The octo-uploadd.py is provided by the sla-slicer-upload recipe:

https://github.com/prusa3d/Prusa-Firmware-SL1/blob/master/sources/meta-prusa/recipes-firmware/sla-slicer-upload/sla-slicer-upload_git.bb

The recipe usually contains links to source code. In this case the source code is located on the gitlab.com:

https://gitlab.com/prusa3d/sl1/project-upload-service

Regarding your interest in Octoprint API, currently there is very little that can be done easily. We are working to introduce interfaces and split the printer control software into components. Once this is done it should be much easier to write an adapter for Octoprint.

njcrawford, 2019-10-09:
Thank you for the pointer! I have an immediate need for the printer and job APIs, so I'll probably hack something together for now and revisit the issue once interfaces are ready to test.

closed 2019-10-09 by njcrawford