prusa3d/Prusa-Firmware-SL1 #2: SL1 can not connect to hidden network

Lichtjaeger, 2019-10-04:
I hide the SSID of my wifi home network. If I try to connect the SL1 I get an error.
If I set my AP to show the SSID, the SL1 connects without a problem. The error is back when I hide the SSID again.

Related to [Gitlab #7](https://gitlab.com/prusa3d/sl1/sl1/issues/7). (still not solved)

port513, 2019-10-05:
This I reported the first day I got my SL1 and they said they are working on it but this is the second FW since release and still nothing have been done about this.

port513, 2019-11-07:
> We are working on firmware version 1.3.0, which will include a new network manager, with which you will could connect to a hidden network.

That's nice to hear since this should be checked in the beginning, Linux dists are know for this issue out of the box.

port513, 2020-01-09:
Still no release that fix this annoying problem, is this firmware still developped?
Quite expensive printer that doesn't get the love from devs as it should.
I have more or less parked my printer for now, the WiFi connection is my most annoying problem except the total reset when flashing the printer.

filipkotoucek, 2020-01-09:
:D ou you quipster. Actually we sent the 1.3.0 to the beta testers today and we plan to go stable next week.

closed 2020-01-15 by Lichtjaeger