prusa3d/Prusa-Firmware-SL1 #3: FW 1.2.3 resets calibration data

port513, 2019-10-05:
This is a horrible way to handle FW update, why on earth is calibration data reset when you flash a new FW?

In manual it says you calibrate once and no more before moving the printer or change FEP.

So please explain why you have choosen to reset calibration data. I spent at least 45 minutes to get that super perfect calibration and now it's gone and ead!

filipkotoucek, 2019-10-07:
We apologize for any inconvenience. The recalibration is mentioned in the previous version 1.2.2 [release notes](https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/v1.2.2) due to print quality improvements. If you wish to calibrate printer as similar as possible, you may export the logs and search for last appearance of following "INFO - HwConfig - AllItems: tiltHeight", and set that number in next tilt calibration. Please take note you will not be able to enter the exact same number. Anyway, in future updates we are going to add notifications to be more clear.

port513, 2019-10-08:
My printer now prints skewed prints, before your force reset it printed so high quality prints. Now I can't make the SL1 print anything at all without distortions on the print.
I even used a zoon camera while calibrating the vat and so on so now my printer is almost useless.

So once again why did you reset without notice?

One other thing that is really annoying, if I calibrate to get that perfect line, notice the numbers and move the vat up and down I can never get to the same number where I had the perfect line. Why is it this way?

Lets say I have a nice even thick line between vat and LCD etc at 4756, I then move up to 5000 and try to get back to 4756 again it's impossible to go to that number again. So how accurate is the calibration really? And how do I fix my issues with skewed prints now? I can't have a clearer line while calibrating the printer so it can't be the calibration that makes all my prints skewed.

Tried to print a cube 20x20x20 mm and the cube isn't squared anymore, but the cube I printed before FW 1.2.3 is perfectly squared. Is it really safe to upgrade FW anymore?

filipkotoucek, 2019-10-09:
We covered this possibility of recalibration in release notes as I stated in the previous comment. We don't support popup notifications at the moment, but we are working on new system of updates where you will see the release notes before you update the machine.

The calibration was reset because the tilt has lower holding torque in certain positions (when motor is not in fullstep position). It resulted on some printers in print quality issues, so the tilt now needs to be calibrated in fullstep. If the previous tiltHeight was not in fullstep position, the printer needed new calibration.

If you have skewed prints it means tilt was not calibrated properly. You can mark the front side of the cube you printed and place it on the table. If the front wall is overhanging, the tilt was calibrated too high. If the front wall has positive slope, the tilt was calibrated too low. Common value of the tiltHeight is usually above 5000.

Yes it is safe to do the FW upgrades. We have no reason to make our product worse. I acknowledge that recalibration might not be the best user experience, but it is designed to take few minutes and as easy as possible.

port513, 2019-11-07:
> If you have skewed prints it means tilt was not calibrated properly. You can mark the front side of the cube you printed and place it on the table. If the front wall is overhanging, the tilt was [...]

Exactly as before you did this reset, now I can't really get any print done perfectly. Done at least 100 calibration now with no good result. Even zoomed in to get the calibration perfectly but still skewed prints every now and then, some prints doesn't show that skewed prints but they are not perfect as before this reset of calibration data. So at the moment my SL1 is a dust catcher, not fun to use anymore. And yes, I tried to go above and below my original values and it still prints bad.

port513, 2019-11-08:
> Exactly as before you did this reset, now I can't really get any print done perfectly. Done at least 100 calibration now with no good result. Even zoomed in to get the calibration perfectly but [...]

The skewed prints might be an error in slicer since other prints seems ok

port513, 2019-11-17:
> The skewed prints might be an error in slicer since other prints seems ok

The slicers seems to make something funny when slicing that object to SL1 since in MK3S it works as it should and other objects is fine in SL1 so this issue can be ignored for now I think.
But please add a note when flashing that calibration data will be reset since it's annoying to have filled up the vat and just realice you have to clean it before you can print again.

closed 2019-12-03 by michalxfanta