prusa3d/Prusa-Firmware-SL1 #4: Print stopped, sl1 became unresponsive and required reboot.

filipkotoucek, 2019-10-07:
Issue created 2.6.2019, moved from https://gitlab.com/prusa3d/sl1/meta-sl1/issues/1

Started a print last night. Didnt start print in the center of the bed, was trying to use a different part of resin tank fep.
https://www.thingiverse.com/thing:3183713
Prusa Orange Tough resin
0.05 layer height
Supports from build plate only
No pad

Model is hollow from resin printing
Print looks about 1/2 way done, and did not complete. The sl1's touch screen went totally black. The power button was flashing on then off.
Touching the screen did not wake up the display.
Pushing the flashing power button did not wake up the screen.
I heard the fan still running, the unit was powered just seems it crashed.
Only choice was to reboot, I was not prompted to restart print after sl1 rebooted.
Heres the log.
[log.CZPX1619X009XC00332.1559484235 (1).txt.gz](https://github.com/prusa3d/Prusa-Firmware-SL1/files/3697750/log.CZPX1619X009XC00332.1559484235.1.txt.gz)

mylife4aiurr, 2019-10-13:
This issue can be closed. It was fixed by newer firmware. Im the original poster on gitlab

filipkotoucek, 2019-10-14:
Thanks. Closing this issue

closed 2019-10-14 by filipkotoucek