prusa3d/Prusa-Firmware-SL1 #6: NEED a Resin Exposure Finder for SL1

mylife4aiurr, 2019-10-13:
Day 1 purchaser of SL1.  

There are countless brands of resin that respond to 405nm light. I thank Prusa for making profiles for some of the other popular brands besides Prusa's own resin. 
Recently I have started trying to use other brands of resin. Currently I have to test one exposure per session on SL1. Testing 1 exposure per session is time consuming and expensive. 
Several other manufacturers have a gcode instruction that test several different exposure times on a single buildplate during a single print session. So a user can find good bottom layer and per layer times quickly and efficiently. We need a similar tool for SL1 - please. 

In the photo Im testing two different exposure times with a white resin on lithophanes, I dont know if the photo is accurate enough but farther testing is required and i'll have to do it one print at a time. Wasting time and resin in the process, having to cleaning up and process each print before I can evaluate it. 
[image]

In this photo I got the x and y dimensions accurate with elegoo green but circles not so much
[image]

mylife4aiurr, 2019-10-13:
[image]
7.0s per layer
7.5s per layer
8.0s per layer

mylife4aiurr, 2019-10-17:
Thanks to cerium136 over on prusa forums
https://forum.prusaprinters.org/wp-content/uploads/2019/10/SL1_Resin_Calibration_Mode.pdf

Found out there is a capability in sl1's firmware. Use at your on risk but him and I have been trying to figure out how to use it. Still would love some official guidance.

mylife4aiurr, 2019-11-22:
Hi

mylife4aiurr, 2019-12-02:
Any response from Prusa on this?

michalxfanta, 2019-12-04:
We prepare this feature with a description of how to use it and with a procedure on how to find correct exposure time for tested resin, in some of the next releases.

mylife4aiurr, 2019-12-04:
Ok that's all I wanted was a response and acknowledgement

X3msnake, 2019-12-04:
https://www.facebook.com/Photonsters/posts/932832370504077

X3msnake, 2019-12-04:
@michalxfanta why did the Prusa team opted out for using standard Gcode?

filipkotoucek, 2020-01-15:
https://github.com/prusa3d/Prusa-Firmware-SL1/issues/8#issuecomment-574888377

closed 2020-01-15 by filipkotoucek

mylife4aiurr, 2020-02-27:
How do we calibrate initial exposure time??
I've asked here:
https://help.prusa3d.com/en/article/sl1-resin-calibration-object_112182

and here also in the comments:
https://youtu.be/_rRCGzKNTxc