prusa3d/Prusa-Firmware-SL1 #7: Ethernet does not work properly

acorderob, 2019-11-09:
My ethernet connection seems to fail on a 1 second interval. The printer shows its IP blinking, and my switch light blinks too from orange (100mb) to off. However WIFI works ok.

Not sure if this is a hardware or firmware problem.

michalxfanta, 2019-11-15:
We are working on firmware version 1.3.0, which will include a new network manager, Please wait for this version.

filipkotoucek, 2019-11-18:
Hey @acorderob. Could you share your ethernet setup in the meantime? What switch/router are you using, cable lengths, CATs, ... Thanks!

acorderob, 2019-11-18:
* Asus RT-AC66U router.
* D-Link GO-SW-24G switch.
* 2 meter CAT-7 cable (i have also tried with an older one, probably CAT-6 but not sure).

acorderob, 2020-01-17:
Just tried the new firmware. It still doesn't work. Attached is the log file. I had no WiFi enabled to avoid polluting the traces. I also made sure the cable connection to my switch worked with another device.

Unrelated: i noticed a line "/usr/bin/octo-uploadd.py[317]: API key changed from" showing my password (i changed that line).

[log.CZPX4119X009XC02169.2020-01-17_17-11-10.txt](https://github.com/prusa3d/Prusa-Firmware-SL1/files/4077702/log.CZPX4119X009XC02169.2020-01-17_17-11-10.txt)

MMetze, 2020-01-17:
Hi!
I had the same issue using cisco managed switches. When I connected the SL1 to a HP ProCurve managed switch, it worked fine. So there seems to be something in the handshake of the more 'advenced' switches, what causes the problem.

acorderob, 2020-01-18:
Except my switch is not managed. So it is either the firmware or my SL1 ethernet hardware is busted (i'm leaning to the latter).

MMetze, 2020-01-19:
I thought the same, contacted support and got a replacement board. But the error persisted. When I switched to the procurve (also managed) it worked fine.
I went into storage and got a Hirschman Mach (managed), a dlink(unmanaged) and some noname 8port (unmanaged) . On the Hirschman ans dlink, the problem persists. Noname worked fine. I think there I something wrong in the handshake handling on the sl1 network firmware. If any logs are needed, let me know.

filipkotoucek, 2020-04-21:
The problem lies down in link layer of the A64 ethernet port. SL1 can't properly setup stable link connection. Fix will come in 1.4.0. We apologize for the inconvenience.

MMetze, 2020-04-21:
Thanks for the feedback!
No worries.

mylife4aiurr, 2020-04-24:
Have been using wifi this whole time I had SL1, just tried to use ethernet port. sl1's loops between plugged in and unplugged, doesnt connect, cant get dchp assigned ip.

Sindarius, 2020-05-06:
> The problem lies down in link layer of the A64 ethernet port. SL1 can't properly setup stable link connection. Fix will come in 1.4.0. We apologize for the inconvenience.

Is there a time frame for a 1.4.0 release? I am experiencing the same connect/disconnect loop on ethernet and was having similar issues with the wifi (not sure if related)

filipkotoucek, 2020-05-18:
I just wanted to let you know this issue is fixed in out recent rc build https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.4.0-rc.1. It basically forces half duplex mode, since there wasn't any other way to fix that on current revision of PrusA64-SL1 board. The drop of transfer speed is around 10 to 15 %. We plan to release the stable build in week or two.

closed 2020-05-18 by filipkotoucek