prusa3d/Prusa-Firmware-SL1 #8: Can the printer run Gcode - I want to be able to port Resin XP v2 to it

X3msnake, 2019-12-04:
https://www.facebook.com/Photonsters/posts/932832370504077

I'm working on Version 2 of Resin Exposure Finder tool, and i am trying to make this available to as much printers as possible.

Can the current firmware do per layer positioning and exposure times?

Can we controll the peel motion like we can on other zip gcode reader printers or at least do multiple exposures without doing peel?

mylife4aiurr, 2019-12-05:
Did you see this:
https://m.facebook.com/groups/1942730525817950?view=permalink&id=2562694483821548

Until prusa releases something official u can use this maybe? I don't if it works for you purpose but I was able to find good exposures for 3 resins prusa hadn't posted profiles for.

filipkotoucek, 2020-01-15:
We just released new firmware https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/v1.3.0. After update you can now easily tune the settings for particular resin. Just download example projects in settings->advanced settings->download examples and print resin calibration object. It is special project for calibration which makes multiple copies of one model over whole print area. By changing the exposure times before print you can setup the time offsets between them and after print finnishes, check which time settings was the best.

closed 2020-01-15 by filipkotoucek

X3msnake, 2020-01-15:
So how do you controll the position of the instances is this done on the file and on prusa or is this done at the machine?

filipkotoucek, 2020-01-20:
The printer itself uniformly positions copies of the object. So the project contains only one object in the center of platform. The number of copies is defined by parameter `calibrateRegions = X` in `config.ini` file in project.