prusa3d/Prusa-Firmware-SL1 #12: Cannot connect PrusaSlicer to SL1

fookatchu, 2020-02-26:
Firmware 1.3.2 (and 1.3.0)
PrusaSlicer 2.2.0-beta

When trying to do a test connection from PrusaSlicer 2.2.0-beta to my SL1 with Firmware 1.3.2 I get the following error message:

```
Could not connect to Prusa SLA: HTTP 404: <html>
<head><title>404 Not Found</title></head>
<body>
<center><h1>404 Not Found</h1></center>
<hr><center>nginx/1.17.0</center>
</body>
</html>
```

Something similar happens with 1.3.0 but the error code is a 401 (Authorization). I don't have the exact error copied before upgrading, sorry. I tried adding the user/pw string to the url, but that also didn't help.

Is the file upload currently broken due to the new webinterface?

fookatchu, 2020-02-26:
Forgot to mention: The webinterface on 1.3.0 and 1.3.2 work as expected. Btw, the new one looks nice!

Area5142, 2020-02-27:
OS used: Linux Mint 19.3, FireFox browser

Have the same problem with SL1 FW 1.3.2 connecting PrusaSlicer (2.2.0 beta) test result is the same. Trying to browse for the printer using the test button results in no printer found.
Accessing the printer via web browser gives the new status screen.

filipkotoucek, 2020-03-05:
You might forgot to fill in the port 8000. So the full path should look like <IP>:8000. The port is also displayed when you hit the "Browse" button. I found that button very helpful, especially when you have to manage multiple SL1.

closed 2020-03-05 by filipkotoucek

fookatchu, 2020-03-11:
Somehow the search through the 'browse' button didn't work for me the last time I checked, but now its working. Might be my network or the new 1.3.3 firmware. 
You are right, port 8000 seems to work. Thanks!