prusa3d/Prusa-Firmware-SL1 #13: Format percent downloaded when updating firmware

devilbiter, 2020-03-05:
When updating the firmware, the "percent downloaded" isn't formatted and has about 16 digits. 
[image]

fookatchu, 2020-03-11:
When it reaches 80% we have zero digits :). As a side note: My Autoupdate to 1.3.3 was only triggered when booting the machine, not when searching for a new firmware through the menu.

filipkotoucek, 2020-04-21:
The issue with decimal points is fixed and will be included in release 1.4.0.

About the update. Since we switched the update mechanism to automatic updates, the "firmware update page" is used in 1.3.3 only for updates from USB. We added the button to force update check on "firmware update page" sou you can check manually during runtime.

All this (and more) will come in 1.4.0 :)

filipkotoucek, 2020-05-18:
I just wanted to let you know this issue is fixed in out recent rc build https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.4.0-rc.1. We plan to release the stable build in week or two.

closed 2020-05-18 by filipkotoucek