prusa3d/Prusa-Firmware-SL1 #15: In 1.3.3 it's impossible to turn off rear fan

ovDarkness, 2020-03-27:
My rear fan is damaged so I print with it disabled.
After update to 1.3.3 I can;t disable it anymore, so my printer is not operational.

filipkotoucek, 2020-04-21:
Thanks for the report. This issue will be fixed in next release (1.4.0).

GBrothers, 2020-05-04:
Oh yes, please fix this issue. I also cant print since I´ve updated to 1.3.3. I replaced the rear fan at all. I connected the SL1 to an external fan via tube to bring the smell out of the room. Thanks! I´m looking forward to be able to use my SL1 again.

filipkotoucek, 2020-05-18:
I just wanted to let you know if you are encountering troubles with rear fan, you can use our recent rc build (1.4.0-rc.1). We plan to release the stable build in week or two.

closed 2020-05-18 by filipkotoucek