prusa3d/Prusa-Firmware-SL1 #16: Currupting the file system on pendrive when deleting files

MMetze, 2020-04-30:
SL1 FW1.3.3
When deleting a file(s) using the UI, the filesystem on the usb pendrive gets corrupted and has to be repaired. All files are lost.

Steps to reproduce:
- Plug in pendrive
- Start printer
- Navigate to Print
- Delete some files
- Start a print
(after the print is done, the printer shuts down)

If the printer gets turned on again, it will not show the files on the pendrive.
When the pendrive is then connected to a PC (windows10) it detects a corrupted file system and after repair, the drive is empty.

filipkotoucek, 2020-04-30:
In the past we encountered only corrupted files on cheap or damaged pendrives. But thanks for the feedback. Could you attach the log from your printer?

MMetze, 2020-04-30:
I used the one provided with the printer. Other than this issue, the drive works fine.
[log.CZPX0320X009XK03046.2020-04-30_11-13-04.txt.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/4557326/log.CZPX0320X009XK03046.2020-04-30_11-13-04.txt.zip)

filipkotoucek, 2020-04-30:
Have you tried with multiple different pendrives? I tried your procedure with multiple flash drives (original from Prusa and other laying donw on my desk) but found no problem. But will investigate it more.

MMetze, 2020-05-03:
I just tried a Kingston Data-Traveler, the effect is the same.
This is a replacement board, maybe there is something else amiss here.

JKinch03, 2020-05-31:
I just had it happen to me yesterday. It's probably something to do with formatting on bootup, allocating space, or with a wrong electrical current flowing through the USB. I have the SL1's original motherboards and the USB that came with it. The files aren't even corrupted, they've just simply been wiped from the memory of the flash drive.

filipkotoucek, 2020-06-08:
It has been a while. We are investigating this issue, but with no results yet (everything works fine here). I was just wondering if your particular setup could have any influence on that. I got few questions:
1. What voltage do you use to power the SL1? 230 or 110 V?
2. What type of USB connector do you have in SL1? You could have just USB extension cord or small PCB screwed  on the metal case.
3. Have you tried to connect flash drive via USB hub and not directly into the printer?
4. Have you anytime experienced same issue when you plugged the flash drive while the printer was already on?
Thanks!

JakoobCZ, 2020-08-07:
Hi,
we have adjusted the behavior of the USB drive in the firmware release 1.4.1 and it should provide protection against data corruption. Download it via OTA or here: https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.4.1

I hope this will solve the issue for you, if not, please let us know.

closed 2020-08-07 by JakoobCZ