prusa3d/Prusa-Firmware-SL1 #19: Files Deleted From USB Drive on Startup

JKinch03, 2020-05-31:
I noticed yesterday that all the files on my Flash Drive had been removed when I turned on the SL1. Usually, I remove the flash drive before turning on the SL1, so I can transfer files to it. This time, I didn't take out the flash drive, because I already had a file on it, and when I went into the print menu, none of the files from the flash drive were showing up. 
**To replicate:**

- Plug in a flash drive with print files on it to the USB port on the SL1
- Turn on the SL1 and wait for boot-up
- Go to Print and you shouldn't see any files appearing from the flash drive
- Remove the flash drive and plug it into a computer to see if there are any files on it

My best guess it that it's probably something to do with formatting on boot-up, allocating space, or with a wrong electrical current flowing through the USB. I have the SL1's original motherboards and the USB that came with it, so it might just be that this was an issue with some of the first production units.

filipkotoucek, 2020-06-08:
It has been a while. We are investigating this issue, but with no results yet (everything works fine here). I was just wondering if your particular setup could have any influence on that. I got few questions:
1. What voltage do you use to power the SL1? 230 or 110 V?
2. What type of USB connector do you have in SL1? You could have just USB extension cord or small PCB screwed  on the metal case.
3. Have you tried to connect flash drive via USB hub and not directly into the printer?
4. Have you anytime experienced same issue when you plugged the flash drive while the printer was already on?
Thanks!

JKinch03, 2020-08-15:
1: 110.
I only connect the USB directly through the USB hub.
No, I have not experienced the same issues while the printer was on and then I plugged the flash drive in.

filipkotoucek, 2020-09-21:
This issue was fixed in https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.4.1 there was a possible race condition in the automounter script we use. We also plan to mount USB as read-only in the upcoming release 1.5.0.

closed 2020-09-21 by filipkotoucek