prusa3d/Prusa-Firmware-SL1 #22: [Feature Request] Dialog - Alternative to slide/swipe

Area5142, 2020-06-23:
In the GUI dialog an action to proceed or shift view often requires a sliding/swipe action - I would like this to operate the machine without needing to swipe but only use buttons.
I know swipe/slide is the common way smart phones operates and an intuitive way of interaction with a touch screen. But when wearing rubber gloves, as I always do around my SL1, this can be a difficult way to operate the SL1 screen. Sometimes, only part of my hand is guaranteed resin-free.
Please provide alternative button actions as a supplement to the swipe.

Another unfortunate common feature of the dialogues is that "Back", "Close", etc are often located at the top of the screen and therefore very difficult to see when standing. I normally operating the machine standing - it's only when I try to see if the print still hangs on the building board that I sit down.
Please consider moving the buttons from the top to mid/low/bottom of the screen.

JohnnyDeer, 2020-07-27:
Hi @Area5142, the swiping on screen is kinda safety feature against ghost touches. So we wont change it at this moment.
About second suggestion. We dont plan to change placing buttons on display any time soon. Thank you for understanding.

Area5142, 2020-07-27:
I understand the arguments, but there is still buttons on the screens and they can still be activated by ghost touches - not all is swipe.
I would like to operate the screen without touching it (as the gloves could be polluted with resin) - Can a tool like a pen for smart phones and tablet be used (I haven't tried it)?
If yes, this could be the solution for people like me with clumsy fingers in rubber gloves... ;-)

JakoobCZ, 2020-08-07:
Hi,
thanks for the suggestions, as my colleague stated the "swipe to action" was introduced in order to prevent the ghost touches on selected screens. Regarding the changes to the UI (buttons position), we will consider it.

The touchscreen is using capacitive technology, therefore any pen with support for this kind of screen should work.

closed 2020-08-07 by JakoobCZ