prusa3d/Prusa-Firmware-SL1 #23: Version 1.4 Bugs

danielkgore, 2020-07-22:
The latest version of the firmware does not seem to check to see if there's enough resin for the print, and never prompted me to add more before the print started.

Most of the way through my first print on Version 1.4, the printer stopped and asked me to add more resin.  I raised the lid and installed the splash guard, and in doing so my finger brushed the screen, nowhere near the "continue" button, and with the lid up the print continued!

There's 3 distinct issues here: removal of a very useful feature (pre-check of enough resin), and two bugs: the whole screen is sensitive to touch rather than just the button, and the print running with the lid up (isn't that a significant safety issue?!?!).

JakoobCZ, 2020-08-03:
Hi Daniel,
we are working on some updates to the resin refill behavior, so we will also check, what you have experienced, thanks for reporting it. Regarding the cover check button, I will investigate asap, the printer should always check for it first.

filipkotoucek, 2020-09-21:
This issue was solved already in https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.4.1

closed 2020-09-21 by filipkotoucek