prusa3d/Prusa-Firmware-SL1 #26: Unable to Update

P-C-R, 2020-10-16:
Hello,

 

i can not update my SL1. Tried updating via WiFi and USB. with 1.4.0 and 1.4.2 everytime it reboots and fails.

Here is my log:

`Oct 16 14:42:58 prusa64-sl1 kernel: Booting Linux on physical CPU 0x0000000000 [0x410fd034]
Oct 16 14:42:58 prusa64-sl1 kernel: Linux version 4.19.81 (oe-user@oe-host) (gcc version 9.1.0 (GCC)) #1 SMP PREEMPT Wed Mar 4 15:12:55 UTC 2020
Oct 16 14:42:58 prusa64-sl1 kernel: Machine model: PrusA64-SL1
Oct 16 14:42:58 prusa64-sl1 kernel: cma: Reserved 32 MiB at 0x000000007e000000
Oct 16 14:42:58 prusa64-sl1 kernel: On node 0 totalpages: 262144
Oct 16 14:42:58 prusa64-sl1 kernel:   DMA32 zone: 4096 pages used for memmap
Oct 16 14:42:58 prusa64-sl1 kernel:   DMA32 zone: 0 pages reserved
Oct 16 14:42:58 prusa64-sl1 kernel:   DMA32 zone: 262144 pages, LIFO batch:63
Oct 16 14:42:58 prusa64-sl1 kernel: psci: probing for conduit method from DT.
Oct 16 14:42:58 prusa64-sl1 kernel: psci: PSCIv1.1 detected in firmware.
Oct 16 14:42:58 prusa64-sl1 kernel: psci: Using standard PSCI v0.2 function IDs
Oct 16 14:42:58 prusa64-sl1 kernel: psci: MIGRATE_INFO_TYPE not supported.
Oct 16 14:42:58 prusa64-sl1 kernel: psci: SMC Calling Convention v1.1
Oct 16 14:42:58 prusa64-sl1 kernel: random: get_random_bytes called from start_kernel+0x94/0x3f0 with crng_init=0
Oct 16 14:42:58 prusa64-sl1 kernel: percpu: Embedded 21 pages/cpu s47576 r8192 d30248 u86016
Oct 16 14:42:58 prusa64-sl1 kernel: pcpu-alloc: s47576 r8192 d30248 u86016 alloc=21*4096
Oct 16 14:42:58 prusa64-sl1 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Oct 16 14:42:58 prusa64-sl1 kernel: Detected VIPT I-cache on CPU0
Oct 16 14:42:58 prusa64-sl1 kernel: CPU features: enabling workaround for ARM erratum 843419
Oct 16 14:42:58 prusa64-sl1 kernel: CPU features: enabling workaround for ARM erratum 845719
Oct 16 14:42:58 prusa64-sl1 kernel: Speculative Store Bypass Disable mitigation not required
Oct 16 14:42:58 prusa64-sl1 kernel: Built 1 zonelists, mobility grouping on.  Total pages: 258048
Oct 16 14:42:58 prusa64-sl1 kernel: Kernel command line: rootwait panic=10 video=HDMI-A-1:D drm.edid_firmware=HDMI-A-1:edid/ls055r1sx04_148.5mhz.bin logo.nologo vt.global_cursor_default=0 console=ttyS0,115200 board_revision=0 stmmac.quirks=1 rauc.slot=B root=LABEL=rootfs.1
Oct 16 14:42:58 prusa64-sl1 kernel: Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes)
Oct 16 14:42:58 prusa64-sl1 kernel: Inode-cache hash table entries: 65536 (order: 7, 524288 bytes)
Oct 16 14:42:58 prusa64-sl1 kernel: Memory: 980740K/1048576K available (7294K kernel code, 804K rwdata, 2360K rodata, 5504K init, 333K bss, 35068K reserved, 32768K cma-reserved)
Oct 16 14:42:58 prusa64-sl1 kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Oct 16 14:42:58 prusa64-sl1 kernel: rcu: Preemptible hierarchical RCU implementation.
Oct 16 14:42:58 prusa64-sl1 kernel:         Tasks RCU enabled.
Oct 16 14:42:58 prusa64-sl1 kernel: NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Oct 16 14:42:58 prusa64-sl1 kernel: GIC: Using split EOI/Deactivate mode
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: Enabling global workaround for Freescale erratum a005858
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: CPU0: Trapping CNTVCT access
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: cp15 timer(s) running at 24.00MHz (phys).
Oct 16 14:42:58 prusa64-sl1 kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Oct 16 14:42:58 prusa64-sl1 kernel: sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Oct 16 14:42:58 prusa64-sl1 kernel: Console: colour dummy device 80x25
Oct 16 14:42:58 prusa64-sl1 kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=24000)
Oct 16 14:42:58 prusa64-sl1 kernel: pid_max: default: 32768 minimum: 301
Oct 16 14:42:58 prusa64-sl1 kernel: Mount-cache hash table entries: 2048 (order: 2, 16384 bytes)
Oct 16 14:42:58 prusa64-sl1 kernel: Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes)
Oct 16 14:42:58 prusa64-sl1 kernel: ASID allocator initialised with 32768 entries
Oct 16 14:42:58 prusa64-sl1 kernel: rcu: Hierarchical SRCU implementation.
Oct 16 14:42:58 prusa64-sl1 kernel: smp: Bringing up secondary CPUs ...
Oct 16 14:42:58 prusa64-sl1 kernel: Detected VIPT I-cache on CPU1
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: CPU1: Trapping CNTVCT access
Oct 16 14:42:58 prusa64-sl1 kernel: CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
Oct 16 14:42:58 prusa64-sl1 kernel: Detected VIPT I-cache on CPU2
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: CPU2: Trapping CNTVCT access
Oct 16 14:42:58 prusa64-sl1 kernel: CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
Oct 16 14:42:58 prusa64-sl1 kernel: Detected VIPT I-cache on CPU3
Oct 16 14:42:58 prusa64-sl1 kernel: arch_timer: CPU3: Trapping CNTVCT access
Oct 16 14:42:58 prusa64-sl1 kernel: CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
Oct 16 14:42:58 prusa64-sl1 kernel: smp: Brought up 1 node, 4 CPUs
Oct 16 14:42:58 prusa64-sl1 kernel: SMP: Total of 4 processors activated.
Oct 16 14:42:58 prusa64-sl1 kernel: CPU features: detected: 32-bit EL0 Support
Oct 16 14:42:58 prusa64-sl1 kernel: CPU: All CPU(s) started at EL2
Oct 16 14:42:58 prusa64-sl1 kernel: alternatives: patching kernel code
Oct 16 14:42:58 prusa64-sl1 kernel: devtmpfs: initialized
Oct 16 14:42:58 prusa64-sl1 kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
Oct 16 14:42:58 prusa64-sl1 kernel: futex hash table entries: 1024 (order: 4, 65536 bytes)
Oct 16 14:42:58 prusa64-sl1 kernel: pinctrl core: initialized pinctrl subsystem
[... 740 lines cut ...]
Oct 16 14:43:19 prusa64-sl1 SLAGUI[300]: qml: Replacing page: home by settings
Oct 16 14:43:20 prusa64-sl1 SLAGUI[300]: qml: New firmware notification "1.4.2"
Oct 16 14:43:20 prusa64-sl1 SLAGUI[300]: qml: Notification.top changed sw_update_available
Oct 16 14:43:20 prusa64-sl1 SLAGUI[300]: qml: Popup window state changed -> sw_update_available
Oct 16 14:43:20 prusa64-sl1 SLAGUI[300]: Notify:  "sw_update_available"
Oct 16 14:43:22 prusa64-sl1 SLAGUI[300]: qml: Notification.top changed
Oct 16 14:43:22 prusa64-sl1 SLAGUI[300]: qml: Popup window state changed ->
Oct 16 14:43:24 prusa64-sl1 SLAGUI[300]: qml: Page <-- support
Oct 16 14:43:24 prusa64-sl1 SLAGUI[300]: qml: Replacing page: settings by support
Oct 16 14:43:26 prusa64-sl1 SLAGUI[300]: qml: Page <-- settings
Oct 16 14:43:26 prusa64-sl1 SLAGUI[300]: qml: Replacing page: support by settings
Oct 16 14:43:28 prusa64-sl1 SLAGUI[300]: qml: Page <-- advancedsettings
Oct 16 14:43:28 prusa64-sl1 SLAGUI[300]: qml: Replacing page: settings by advancedsettings
Oct 16 14:43:30 prusa64-sl1 SLAGUI[300]: qml: Emitting {"page":"advancedsettings","id":"exportlogstoflashdisk","pressed":true}
Oct 16 14:43:30 prusa64-sl1 SLAGUI[300]: qml: Emitting {"page":"advancedsettings","id":"exportlogstoflashdisk","pressed":false}
Oct 16 14:43:30 prusa64-sl1 SLAGUI[300]: qml: Page <-- wait
Oct 16 14:43:30 prusa64-sl1 SLAGUI[300]: qml: Replacing page: advancedsettings by wait
Oct 16 14:43:30 prusa64-sl1 nohup[298]: /run/media/root/sda1/log.CZPX4819X009XK02774.2020-10-16_14-43-30.txt.xz
Oct 16 14:43:30 prusa64-sl1 systemd[1]: run-media-root-sda1.automount: Got automount request for /run/media/root/sda1, triggered by 410 (bash)
Oct 16 14:43:30 prusa64-sl1 systemd[1]: Mounting /run/media/root/sda1...
Oct 16 14:43:30 prusa64-sl1 systemd[1]: Mounted /run/media/root/sda1.
Oct 16 14:43:29 prusa64-sl1 systemd-timesyncd[269]: Synchronized to time server for the first time 176.9.18.143:123 (pool.ntp.org).
Oct 16 14:43:33 prusa64-sl1 nohup[298]: Data from the specified boot (-39) is not available: No such boot ID in journal
Oct 16 14:43:33 prusa64-sl1 nohup[298]: Data from the specified boot (-38) is not available: No such boot ID in journal
Oct 16 14:43:41 prusa64-sl1 systemd[1]: systemd-hostnamed.service: Succeeded.`

P-C-R, 2020-10-19:
@filipkotoucek any Idea?;) Best

P-C-R, 2020-10-22:
I can downgrade to 1.3.2  and upgrade to 1.3.3 but every update to 1.4.x fails. tried the RC Versions too.

P-C-R, 2020-10-22:
[log202022.txt](https://github.com/prusa3d/Prusa-Firmware-SL1/files/5421333/log202022.txt)

here my log from today. I tried to downgrade upgrade factory reset etc...

P-C-R, 2020-10-22:
another Information. The downgrade to 1.3.0 always fails in the menu. "Update failed"

Sometimes when i try to update to 1.4.2 after rebooting a message appears with "Update failed. Booted from diffrent slot. Please update ASAP" or something like this.

JakoobCZ, 2020-10-22:
Hi, it seems this requires a deeper analysis of your hardware, can you please write me an email at info@prusa3d.com and include my name (Jakub Dolezal)? Thanks

P-C-R, 2020-10-22:
@JakoobCZ  Done!

Best

P-C-R, 2020-10-26:
@JakoobCZ till now i got no mail from you. Best

filipkotoucek, 2020-12-18:
What is the status of this issue @P-C-R ?

P-C-R, 2020-12-18:
A replacement board fixed the issue. Old board is back @prusa

closed 2020-12-18 by P-C-R