prusa3d/Prusa-Firmware-SL1 #29: [Request] Tune LED power

tehrealape, 2021-02-02:
Hey there SL-1 developers,

Due to a recommendation from your customer service team, I am opening this "issue".

I am a PhD candidate from the netherlands, who previously already worked on a Prusa i3 MK2 bioprinter conversion. Next I want to use the SL-1 for the same purpose and have converted the existing printer into a working prototype for biofabrication applications.

However, I have come to a limit optimizing the parameter of the resin-mixture used in such scenarios. A remaining parameter I need to tweak is actually the energy emission from the LED. My measurements have shown me that the printer can deliver about 0.1mW/cm² or 1W/m² of UV light directly at the LCD, which was confirmed by your customer service so far.

My question is simple: Can you help me out on increasing the irradiance level by the LED based on a firmware solution? Would it be an option to make this adjustable (e.g. go from "1W/m2" to "2W/m2" mode) ?

Looking forward for an reply,
Nils

filipkotoucek, 2021-02-08:
Hello @tehrealape. If custom FW is considered (which will void the warranty), there is only UV PWM value https://gitlab.com/prusa3d/sl1/a64-fw/-/blob/master/sl1fw/configs/hw.py#L143 which can be tuned. The common value should be set around 200 to get the required intensity. This value can be changed up to 250 in FW.
But I doubt it will give you twice the irradiance on the display. Current display passes through only few % of the power of the LED. It will be good to use for example monochromatic display instead which passes through few tens of %.

JakubSilar3D, 2021-03-02:
Hi @tehrealape , hope the advise above helped. I am closing this issue on that assumption. Feel free to come back if you need more info.

closed 2021-03-02 by JakubSilar3D