prusa3d/Prusa-Firmware-SL1 #30: [Request] Tower homing check before printing

MarinaOr, 2021-02-13:
I am trying to unable the tower homing check before printing. My intention is to print without the printing platform for tissue engineering purposes (I do research in my university; I disassembled the tower from the printer). I do not know a lot about programming, so I would really appreciate your help and guidance for this matter. Thank you in advance!

JakubSilar3D, 2021-04-12:
Hi Marina. Not sure what you are trying to achieve. Can you, please, elaborate? What do you need from the SW change?

MarinaOr, 2021-04-13:
Hey!

At the end I could manage to get what I needed it to do from the hardware.
Anyway, thanks for your help!

Kind regards,
Marina

JakubSilar3D, 2021-04-13:
That is good news.

closed 2021-04-13 by JakubSilar3D