prusa3d/Prusa-Firmware-SL1 #32: cannot read from remote-api repository

venny166, 2021-04-08:
The bitbake recipe in Prusa-Firmware-SL1/sources/meta-prusa/recipes-firmware/sl1fw-api/sl1fw-api_git.bb points to prusa3d/SL1/remote-api on Gitlab, which cannot be read from (only an empty folder is visible). Error 403 is returned when trying to read with SSH.
Only remote-api and meta-sl1 are inaccessible. What could be the reason? Surely I am not the only user who tried to build his own customized firmware version.
Any pointers are welcome.

JakubSilar3D, 2021-04-09:
Hi @venny166, looks like something happened to the rights of the linked project. We are looking into it.

JakubSilar3D, 2021-04-09:
It the access rights where reallly wrong. The access to sl1fw-api should be OK (public), now. We will test it, but you should be good to go and your verification would help.

JakubSilar3D, 2021-04-09:
Fixed and tested. Thanks for letting us know.

closed 2021-04-09 by JakubSilar3D