prusa3d/Prusa-Firmware-SL1 #34: Many errors in the web interface.

flandaluce, 2021-05-08:
When you access the printer control via IP, it does not correctly indicate a lot of data, remaining time, printing time, host name, remaining resin, etc ... and the print image does not appear.

I am attaching a screen.

[image]

JakubSilar3D, 2021-05-08:
Hi. Thanks for your attention. Just to clarify, what firmware version are you using?

flandaluce, 2021-05-08:
Version 1.5.0

JakubSilar3D, 2021-05-09:
These are known issues in 1.5.0. The Prusa Link (new name for Prusa Local connect) web interface will be much better in 1.6.0 which is due in several weeks. You can get it earlier by joining the beta tester group. 
But the 1.6.0 is a big rework of the firmware internals, so all the usual beta disclaimers will apply.

JakubSilar3D, 2021-07-19:
Most of the things you mention are fixed in 1.6.0 which has been just released.

closed 2021-07-19 by JakubSilar3D