prusa3d/Prusa-Firmware-SL1 #35: Updating the firmware to 1.6.0 lost 24 hour setting

Area5142, 2021-07-15:
Just updated the firmware on my SL1 over the net from 1.5.0 to 1.6.0. Everything went well except that the display lost its previous 24 hour setting replaced it with AM/PM notation.
Manually setting time is presented in 24 hour format but the display keeps using AM/PM.

Chosen language is English and locale is Denmark/Copenhagen.

I found no way to change from AM/PM to 24 hour format in printer settings.

malcinator, 2021-07-17:
I have the same issue. 

Time is set by NTP from internet. No option for 24hr time or visible way to change it back to 24hr.

JakubSilar3D, 2021-07-19:
The default *nix behaviour is like that. This is a change, which was requested by other users. Now the 24hr <-> AM/PM is set to default for the given language/locale, while previously it was "24hr is good for everyone". We are considering to make it user selectable. 
This does not depend on whether your printer clock is synced with from the network.

Area5142, 2021-07-20:
@JakubSilar3D I am confused.
Local language default for the given Europe/Copenhagen region is 24 hour display, even if English is selected for language. Likewise the date display format is in the form day-month-year.
I think this is to be considered as a bug.

malcinator, 2021-07-20:
I’m feeling that bugs are being portrayed as features and what were features now being portrayed as bugs. What every they are this release feels more incomplete and rushed out the door, 1.5.0 was much the same.

Area5142, 2021-08-18:
Just installed version 1.6.1 and it is now possible to select between _native/12 hour/24 hour_ format for time and the display is updated right away.
The date format still seams to be stuck in _mm/dd/yyyy_ format where the correct format for Denmark is _dd/mm/yyyy_ and should be defined when the country is selected.

Prusa-Support, 2022-05-06:
Thank you for the report, it is not a mandatory feature but we understand why it should be offered. This is noted and hopefully, we can add this option anytime soon

Sarah Kapodistria
Prusa Research

closed 2022-05-06 by Prusa-Support