prusa3d/Prusa-Firmware-SL1 #36: Keyboard Input Still No Working

malcinator, 2021-07-17:
Unpluged the ethernet so I can revert back to Wifi as keyboard input was broken in the 1.5.0 release. I was hoping it would be fixed in this release, its not!

Trying to enter my WiFi password is impossable using a working USB keyboard.

JakubSilar3D, 2021-07-19:
Hi. We do not support an external keyboard (or, more generally, anything else than usb storage on the front panel. This is intentional. It did partly work in the previous versions, but that was a bug. Sorry for the inconvenience, but it is meant to be that way.

closed 2021-07-19 by JakubSilar3D

malcinator, 2021-07-19:
As per https://www.help.prusa3d.com/en/article/internet-connection-sl1_1981#wireless-connection

Wireless connection

The SL1 3D printer has a built-in Wi-fi network card compatible with 2.4GHz networks. To open the Wi-Fi network browser, go to Main Menu -> Settings -> Network -> Wi-Fi. A list of all available networks will be displayed. Select your Wi-Fi network and connect to it by typing your password using the on-screen touch keyboard. 

You can connect a USB keyboard to the front USB port in case the on-screen keyboard is too small.

This is a “Feature” according to the knowledge base article. I don’t believe this to be a bug as you call it. When 1.5.0 was released and I reported it it was going to be fixed in the next release. Now it’s a bug?