prusa3d/Prusa-Firmware-SL1 #37: No way to back out of WIFI Password input.

malcinator, 2021-07-17:
Tried to enter my very complex WiFi Password using the on screen keyboard. Having tried and decided to re-run the ethernet cable back to the SL1 I found I cannot back out of the onscreen input screen, the cancel button beeps but does nothing. Only powering down the SL1 has been my only option.

Is there anyway to revert firmware back to a older release?

JakubSilar3D, 2021-07-19:
Yes, settings->Firmware->Switch To Version should bring you to the previously installed version.

malcinator, 2021-07-21:
That would then take me back to 1.5.0. How would I for example reinstall the 1.4.2 firmware?

Prusa-Support, 2022-05-24:
Hello, 
thank you for the report! We tested this and it should be an obsolete issue by now. 
If you still keep getting a similar error, please open a new issue here or send us an email to info@prusa3d.com and we will review the problem

Sarah Kapodistria
Prusa Research

closed 2022-05-24 by Prusa-Support