prusa3d/Prusa-Firmware-SL1 #38: Cannot change rear fan speed

csuttichujit, 2021-07-17:
Apparently, I can no longer change the rear fan speed after updated to 1.6.0.

The issue with adjusting the rear fan speed began from when I updated to 1.5.0. The issue was whenever I adjusted the rear fan speed, the blower fan was adjusted instead. Thus, leaving the rear fan speed unchanged. I informed Prusa, but that was it. Moreover, I was instructed to downgrade to 1.4.2 which I did and everything worked fine again.

Now for the 1.6.0, when I adjusted the rear fan speed to 2500 RPM, the rear fan speed stayed at 1800 RPM as per I set it on fw 1.4.2. However, the blower fan was adjusted down to 2500 RPM instead. Those mentioned numbers were from Prusa Link.

To point out, the adjustable range of rear fan speed is now off-3300 RPM and not off-5000 RPM as it used to be. 3300 RPM is the max of blower fan right? So maybe the adjuster was wrongly mapped since 1.5.0?

[image]

JakubSilar3D, 2021-07-19:
Thank you for spotting that. We will look into in and handle in the next bugfix version.

csuttichujit, 2021-07-19:
> Thank you for spotting that. We will look into in and handle in the next bugfix version.

By the way, my SL1 is now bricked right when I clicked “yes” to downgrade from 1.6.0 to 1.4.2. I already informed Prusa about this and they’re investigating. I figured this issue is off topic, but I can give you more info if you’d like.

malcinator, 2021-07-19:
That is worrying as I have asked about the possibility of reverting to another version of firmware. I have ordered the SL1S upgrade but the 1.6.0 firmware was not what I was expecting. So dilemma, forgo the upgrade and change firmware or hope 1.6.1 is released soon.

csuttichujit, 2021-07-20:
> That is worrying as I have asked about the possibility of reverting to another version of firmware. I have ordered the SL1S upgrade but the 1.6.0 firmware was not what I was expecting. So dilemma, forgo the upgrade and change firmware or hope 1.6.1 is released soon.

I changes back and forth between 1.4.2 and 1.5.0 quite a few times upon testing to make sure the fan issue wasn't caused from my end. It worked fine every time up until I did a 1.6.0 back to 1.4.2. Maybe it was because I jumped down 2 fw at once? but I didn't have 1.5.0 though since I downgraded to 1.4.2 prior to 1.6.0 update.

Anyway, let's hope they have a solution to this quickly. I need to print something and I rarely used my SL1, so bricking like this is a bit of a bummer... but I get it and I'm willing to cooperate in any way to speed things up.

malcinator, 2021-07-20:
I have tried myself to revert back to 1.4.2 from 1.5.0 but it just errored out with something, I forget now what. I guess I use my SL1 like you do. There are some prints I can only get done on my SL1 and as such it’s not used every day.

csuttichujit, 2021-07-20:
> I have tried myself to revert back to 1.4.2 from 1.5.0 but it just errored out with something, I forget now what. I guess I use my SL1 like you do. There are some prints I can only get done on my SL1 and as such it’s not used every day.

[image]
Luckily your SL1 can still recover from the error. I had this screen right when I clicked yes (or a split second right before because I didn't hear any beep sound) to downgrade and I was like oh man that's a kernel panic then it froze at the Prusa logo.

malcinator, 2021-07-20:
Well I’m hoping that the keyboard “bug” as it’s being called now gets fixed soon and I won’t have to restore the 1.4 firmware. I hope that you get a new mainboard and a bottle of resin to say sorry  for the trouble you have had.

csuttichujit, 2021-07-20:
> Well I’m hoping that the keyboard “bug” as it’s being called now gets fixed soon and I won’t have to restore the 1.4 firmware. I hope that you get a new mainboard and a bottle of resin to say sorry for the trouble you have had.

Lol. Funny thing is this happened 2 days prior to the end of my SL1's warranty, still I'm happy they're trying to figure this out for me. By the way, what's the keyboard bug?

malcinator, 2021-07-20:
You used to be able to plug a keyboard in to the front USB port to enter the WiFi password. It was broken in 1.5.0. Now in 1.6.0 I’ve been told it’s a bug and never worked as expected contrary to the Knowledge Base article on entering the password using a USB keyboard. That is the reason I would like to revert back to 1.4.2 so I can enter my password and remove the extremely long CAT5 cable that runs through the apartment.

csuttichujit, 2021-07-20:
> You used to be able to plug a keyboard in to the front USB port to enter the WiFi password. It was broken in 1.5.0. Now in 1.6.0 I’ve been told it’s a bug and never worked as expected contrary to [...]

Why can't you enter your password with the on-screen keyboard instead?

malcinator, 2021-07-22:
The screen is to small for my large fingers. The password is rather long with a mix of everything. I have tried more than a few times to enter it. Maybe I could try using a stylus or something.

Prusa-Support, 2022-05-25:
Hello, 
the rear fan speed was fixed in one of the previous versions. At least it does work on 1.6.6. and 1.7.0 beta, I hope you can also confirm. 

About the Bricked printer, we apologize for the trouble. This could only happen if you're on Version 1.3.0 - 1.4.2 and you intend to update to 1.6.0. and back to the previous version. There is a fix based on the behavior of your printer and I will be happy to guide you via email if any of you still have that issue - info@prusa3d.com

Using a Keyboard for the Password should be again possible to use from version 1.7.0 :) 

Sarah Kapodistria
Prusa Research

closed 2022-09-12 by filipkotoucek