prusa3d/Prusa-Firmware-SL1 #39: Resin sensor consistently reads too high

andrewluebke, 2021-07-19:
After upgrading to 1.6.0, on all of my printers I have to disable the resin sensor because it claims that there is 230-240ml of resin in the tank.  They are filled up to the 100% mark which used to translate to about 160ml.

JakubSilar3D, 2021-07-19:
Hi Andrew,
can you please provide the logs (a typical one)?  Thank you.

JakubSilar3D, 2021-07-19:
Andrew,
1/ Thank you. This looks like a bug, we will start looking into it asap.
2/ Are you OK with switching the resin sensor off as a temporary workaround?
3/ Could you kindly try to run the resin sensor part of the selftest an provide us the log? (Sorry, for not asking you before.)
I hope to get this right in the next bugfix release (no promises, but rather weeks then months).
_Note that our support is dedicated to help you with official releases, while github issues should be focused on community development. (There is no problem in continuing here with your sensor issue, just a general remark. We are small team of developers, so we cannot provide consistent response times, e.t.c.)_

andrewluebke, 2021-07-19:
The resin self test is at Jul 19 13:29:25, I messed up the tower position check afterwards by not turning the platform but I don't think you care about that part.  If you do let me know and I'll do it again.

[logs 2.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/6843770/logs.2.zip)

filipkotoucek, 2021-07-21:
Dear Andrew,

I was looking into your logs and there was nothing wrong. But I found out where the issue probably is. 

In 1.6.0 we introduced a more precise resin sensor measurement algorithm BUT there is a different bug. Tank is not properly levelled when a user is asked to pour resin in. If the user pours resin up to 100 % mark in this situation (tank in unknown hanging position), it means the volume is above maximum. So technically, the error you are getting is correct, but we need to level the tank before the printer asks for resin.

Fix for this bug is already merged in master https://gitlab.com/prusa3d/sl1/a64-fw/-/commit/3ca83f7be7b62f66592d57a4ade522775eea747d and https://gitlab.com/prusa3d/sl1/a64-fw/-/commit/e05ff4c4e2afe4489ca13fcc1a43c6150e7dd373. So 1.6.1 will solve your problems.

Thanks for your cooperation and for notifying us about this issue.

Prusa-Support, 2022-05-06:
Closed - Fixed

closed 2022-05-06 by Prusa-Support