prusa3d/Prusa-Firmware-SL1 #40: Prompt before updating firmware?

malcinator, 2021-07-19:
Would it be possible in future to prompt before updating rather than just downloading the firmware and updating?

JakubSilar3D, 2021-07-19:
Can you, pls describe, what happened in more detail and optionally provide the log? This seems to be a bug - in usual use cases there is no update without a prompt.

malcinator, 2021-07-19:
Sure. I powered on my SL1 and when it connected to the network here it immediately downloaded the upgrade and installed it. I don’t remember being asked to proceed. I would revert to 1.5.0 and go back through the process but I’m now concerned that the firmware downgrade will brick the printer.

malcinator, 2021-07-23:
Downgraded firmware. Powered off. Powered back on and notification to upgrade appeared. So no issue

closed 2021-07-23 by malcinator