prusa3d/Prusa-Firmware-SL1 #42: Error when uploading from PrusaSlicer when Printer is printing

andrewluebke, 2021-07-19:
Starting with 1.6.0 when I upload a new print to the printer when the printer is running I get the following error in PrusaSlicer:

Error uploading to print host:
HTTP 409: #10506: Cannot run this action, wait until the printer finishes the previous action.

However the print has been uploaded and if I try to upload again I get the error about file already existing.  I'll upload a log when my current print is finished.

Area5142, 2021-07-20:
Got the same error just displaying uploaded prints - printer was idle and just turned on:

[image]

The error is repeated and fills the screen.

andrewluebke, 2021-08-02:
Any progress on this one?  It's pretty annoying.  I can confirm too that this happens if you upload multiple files without actually printing anything.