prusa3d/Prusa-Firmware-SL1 #43: Printers report UV LED RPM issues

andrewluebke, 2021-07-20:
After upgrading to 1.6.0 all my printers (5) report error 10714, expect overheating.  The website reports a speed of 2500rpm (not varying) and a temperature of ~47-49 degrees C on the UV LED's, so think the fan is actually turning.  It's unlikely that all of the UV fans failed at the same time so I think it's another firmware bug.  The only error about it I can see in the log is:

Jul 19 00:04:45.188494 SL1-2.customccc.com SL1FW[307]: ERROR - sl1fw.pages.base - Detected fan failure: ['UV LED fan']
Jul 19 00:04:45.196908 SL1-2.customccc.com touch-ui[301]: qml: Trying to push an error:  {"code":"#10714","failed_fans_text":"UV LED fan","name":"e10714.ExpectOverheating","text":""}

andrewluebke, 2021-08-02:
This issue may have been from other issues as I haven't had it since reporting the problem so I am closing the case.

CountMurphy, 2021-12-07:
Seems fixed for me in 1.6.3

andrewluebke, 2021-12-08:
Yeah, I haven’t seen it for quite a while now.

Prusa-Support, 2022-05-06:
obsolete issue and will be closed - thank you, guys!

closed 2022-05-06 by Prusa-Support