prusa3d/Prusa-Firmware-SL1 #49: Resin Calibration Object for SL1S

Undermeadow1, 2021-08-31:
There are no resin calibration objects after downloading the examples in the new SL1S Speed version.

andrewluebke, 2021-08-31:
Yes I saw the same thing.  I wanted to calibrate with some exotic resin and couldn't.

achingbrain, 2021-11-16:
Not in `v1.6.3` either.

@filipkotoucek @zavorka @vladamatena any ideas when the ability to calibrate new resins will be added?

Jerzinek, 2021-11-18:
Hi, we are working on this feature fix. I believe it is a matter of several weeks. Please stay tuned.

VavoSam, 2021-12-13:
Hi, when will be this feature ready?

Area5142, 2021-12-22:
Is Resin Calibration Object part of the newly released 1.6.4 version?

Jerzinek, 2021-12-22:
Not yet, Unfortunately. We are still working on this feature fix.  It
should be released hopefully very soon after the New Year. Apologize for
the inconvenience.

Hauzman, 2022-01-17:
Hi is there a way to calibrate rasin until you release the official one.

Undermeadow1, 2022-01-18:
Yes there is unofficial and seems to work well:
[Resin Calibration SL1S.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/7887882/Resin.Calibration.SL1S.zip)

Hauzman, 2022-01-18:
> Yes there is unofficial and seems to work well:
> [Resin Calibration SL1S.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/7887882/Resin.Calibration.SL1S.zip)
> 

Thank you I will try it out this day's and let you you know.
I need to calibrate some biocompatible materials

Hauzman, 2022-01-21:
> Yes there is unofficial and seems to work well:
> [Resin Calibration SL1S.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/7887882/Resin.Calibration.SL1S.zip)
> 

@Undermeadow1 I can confirm it's working I will make also for the flex resin i use [image][image]

Prusa-Support, 2022-05-06:
Hello Guys, 

thank you very much for your patience! 
The Resin Calibration Objects are about to be in the Stable version, you can already use this feature in the [1.7.0-beta.4] and also try the Resin tank cleaning feature, your feedback on this will be appreciated. Please open a new issue if there is anything as this topic will be closed for now

Sarah Kapodistria 
Prusa Research

closed 2022-05-06 by Prusa-Support

achingbrain, 2022-05-16:
I've just tried the calibration print in the beta firmware, it works well, thanks so much!