prusa3d/Prusa-Firmware-SL1 #57: Can't delete a (bad) file from the printer

JPElectron, 2022-03-03:
If a file uploaded to the printer is "bad" there is no way to delete it since the file never loads.

Additional information in the forum post: https://forum.prusaprinters.org/forum/original-prusa-sl1-hardware-firmware-and-software-help/cant-delete-a-file/

Prusa-Support, 2022-04-22:
Hello, 
thank you for the report! Just to be sure, you're going to the print file overview and you try to swipe on the print you would like to delete but it offers no delete option but for other successful uploads it does? 
If that's the case, could you please send us a short video of the problem to the official tech support email and attach the log file as well? 

Best regards

JPElectron, 2022-05-03:
Video showing how I can't delete bad files...
https://www.youtube.com/watch?v=sSIbl8vRTLc

Also not possible from the printer screen.

Prusa-Support, 2022-05-06:
Thank you for the video! Unfortunately, this is connected to PrusaLink and not directly to the printer.
We tested the last FW versions and it seems to work for us.
Please send us more information about how a file can't be deleted from the printer, preferably via email at info@prusa3d.com with the log file attached and you can ask me directly, I will be happy to help 

Sarah Kapodistria
Prusa Research

closed 2022-05-06 by Prusa-Support