prusa3d/Prusa-Firmware-SL1 #61: Support for WPA2-Enterprise (and by extension WPA3-Enterprise)

Feerip, 2022-05-10:
Hi, are there any plans for adding 802.1X enterprise network support? I believe NetworkManager does support it already, but I can't seem to find any way to input a username in addition to a password for the wifi settings on the SL1Ss we have here. (802.1X needed because the printers are in our makerspace on university campus, there's no WPA/WPA2-PSK networks here, and ethernet is not an option)

Prusa-Support, 2022-05-25:
Hello,
yes, it would be possible to add WPA3 and unsecured networks in future versions. 
For the time being, we are not planning any other protocols unless there are explicit requests in case of SLA farms with at least 10 or more machines

Sarah Kapodistria 
Prusa Research

Feerip, 2022-05-25:
Hi, thank you for the response. I think there was a misunderstanding, and I apologize if I wasn't being completely clear. What I was inquiring about was 802.1X, which is used in conjunction with all three protocols WPA/WPA2/WPA3. Specifically, our university campus uses WPA2-Enterprise, instead of WPA2-PSK, which is still using WPA2, but instead of just having a password (PSK = pre-shared key), it requires both a username and password, for RADIUS based authentication. 

That being said, our makerspace is mostly using MK3S (~20 printers) and only has 4 SL1S printers, and I completely understand if you can't dedicate resources to adding enterprise network support just for us.

However I would like to ask, do you accept 3rd party code contributions? I would be happy to write the code myself if contributions are accepted :) 

Thanks!

BPaton, 2024-03-20:
Can I up vote this please? I am using a MK4 in a university environment where both wifi (WPA2) and ethernet are protected via 802.1x RADIUS authentication. The alternative I think is going have to be a RPi (with a camera for the enclosure) with an additional wifi dongle to have it become a wifi hotspot and let the RPi connect to the uni network and connect the MK4 to the created hotspot. Although I think this might violate the terms of use of the uni's IT services.