prusa3d/Prusa-Firmware-SL1 #64: Feature Request: Disable Tilt Mechanism for "Mechanical Models"

achingbrain, 2022-06-19:
Cross posting this from https://github.com/prusa3d/Original-Prusa-SL1/issues/9 - that repo hasn't had a commit in years, no idea if it's still used for anything.

---

Due to the physics of the tilt mechanism, mechanical parts can be warped as they are pulled unevenly and adhere to the bed for a short time before releasing.

Please add the ability to disable the tilt mechanism on a per print basis to allow for perpendicular delamination from the film via the Z axis.  I believe this will improve quality of mechanical prints.

I suppose this would also require a Prusaslicer update to allow a box to be checked when slicing for the SL1.

Edit: I adding a few examples of people with the issue.
[forum.prusaprinters.org/forum/original-prusa-sl1-hardware-firmware-and-software-help/tilt-and-warped-mechanical-prints](https://forum.prusaprinters.org/forum/original-prusa-sl1-hardware-firmware-and-software-help/tilt-and-warped-mechanical-prints/)
[forum.prusaprinters.org/forum/original-prusa-sl1-general-discussion-announcements-and-releases/lumpy-bottoms-err-tops](https://forum.prusaprinters.org/forum/original-prusa-sl1-general-discussion-announcements-and-releases/lumpy-bottoms-err-tops/)
[forum.prusaprinters.org/forum/original-prusa-sl1-general-discussion-announcements-and-releases/part-defections](https://forum.prusaprinters.org/forum/original-prusa-sl1-general-discussion-announcements-and-releases/part-defections/)

Prusa-Support, 2023-01-02:
Hello, 
thank you for your suggestion! We put this on the list for the future as soon as there is more time for testing we will hopefully be able to offer this feature also

murk-sy, 2024-06-07:
Should be resolved with [1.8.0-beta.0](https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.8.0-beta.0)