prusa3d/Prusa-Firmware-SL1 #67: Printer LCD stopped working after 1.7.0 final update (PRUSA logo not showing in self-check)

F4r0kh, 2022-07-04:
Great job guys... After the update to 1.7.0 final the monochrome LCD stopped working. Was surpriced I didn't hear the pealing sound of the first few layers on my SL1S. Tried cleaning the FEP foil via the new feature, nothing pealed off, poured the resin back to the bottle and the FEP foil was spotless. So I ran a self-check and when it was time to check the display, the PRUSA RESEARCH logo did not show up. After rebooting 3x and downgrading back to the previous stable version the selfcheck went fine and the printer is printing again.

murk-sy, 2022-07-06:
I've had issues installing beta4 and I went to verify this as I just upgraded, but the screen seems to work fine.

Are you using 1.6.6 now and did you attempt to install 1.7.0-beta4 before?

F4r0kh, 2022-07-06:
Yep, went from 1.6.6 to 1.7.0 and back. Have tried to install beta 4 a day before and almost bricked my SL1S, it got stuck on initializing after the update. Anyway yesterday I tried to install the 1.7.0 final again (from the same exact file) and it went fine and the display is working, so no idea why it was acting up the last time I installed it.

Prusa-Support, 2022-07-11:
Hello, 

thank you for sharing this error, we did not see it in any version so far. Could you please send us the log file to info@prusa3d.com or share the upload code? 
We would appreciate more info about the details and perhaps we can find the cause for this within a short time. 
Bricking the printer can happen but there is always a way to fix it. Our Technical support will be happy to guide you through it in case it happens 

Sarah Kapodistria
Prusa Research