prusa3d/Prusa-Firmware-SL1 #68: High Viscosity

calebhelios, 2022-07-14:
High viscosity tilt speed is not able to be edited in the slicer. Even though it is set to 30 seconds, it completes the tilt in around 15 seconds which is too fast for high viscosity resins, so the printer stepper motor skips steps.

xsoriginal, 2022-08-15:
Unfortunately none of the profiles can be edited in slicer currently. The values in the slicer are just to help it give accurate timings.

I really think this needs changing asap. We should be able to adjust print settings like this, just as every cheap Chinese printer can dialled in perfectly.

Prusa-Support, 2022-08-16:
thank you for your feedback. We will review the settings with our developers

murk-sy, 2024-06-07:
Should be resolved with [1.8.0-beta.0](https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.8.0-beta.0)

vadimmarkelov, 2024-10-14:
Looking forward for stable release of 1.8.0