prusa3d/Prusa-Firmware-SL1 #71: Custom profiles

xsoriginal, 2022-08-15:
Is there a way we can edit the timings for the lift/retractions fully?
Every cheap Chinese printer gives full control over speeds and timings and neither Faster, Slower or High-Viscosity are perfect for my resin.
It would also be fantastic if we can disable tilt and use the new vertical lift motion from the high-viscosity profile instead to avoid warping on mechanical parts.

A bit more detail. My resin is viscous but the Slower profile is far slower than I need it to be. I'd like to be able to use the Faster profile but with a 1.0s wait time upon retraction. Without this, I experience serious blooming issues with my resin.

Is this something that can be changed in the firmware?

Prusa-Support, 2023-01-02:
Hello, 
thank you for your request! 
This was presented as a feature point to the development and hopefully, we can offer it in the near future. 
Depending on what your piece looks like, it's best to also add a small drainage hole close to the platform to release the pressure. I understand if this is no option for you, we will try to add more settings in the future.

murk-sy, 2024-06-07:
Should be resolved with [1.8.0-beta.0](https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.8.0-beta.0)

xsoriginal, 2024-06-07:
Great to hear, thanks