prusa3d/Prusa-Firmware-SL1 #72: Incorrect resin estimation

GitPetri, 2022-09-09:
The attached project shows the following after slicing:
Sliced Info
Used materials (ml): 92.70
- objects 89.95
- supports and pad 2.84
When initializing the print with SL1S (latest firmware) the printer asks to fill resin to 90% mark.
And when filling resin to ~100% the printer warns that the amount of resin is not probably enough.
When the printing starts, SL1S UI shows Remaining resin 160 ml

Now with my math skills 160/92.70 ~= 172% i.e the amount of resin should be enough.
Which one is incorrect
- PrusaSlicer (2.5.0) estimation
- SL1S claim about 160 ml
- SL1S claim of non-adequate amount of resin
?
[V2 barrel front.zip](https://github.com/prusa3d/Prusa-Firmware-SL1/files/9532655/V2.barrel.front.zip)

Prusa-Support, 2023-01-02:
Hello,
It is possible that something is touching your printer when measuring, it can be a paper towel or a cable etc.
This lowers resistance and values less material.
Often the resin level is no longer accurate if some resin was spilled and went between tilt and frame. 
Please check and if the performance is not fixed please write to info@prusa3d.com, our support team would be happy to check it further

GitPetri, 2023-01-03:
Nothing touching the printer, it stands alone on a table and only thing touching is is the power cord (of course). Neither am I aware that I'd spilled into the printer. I'll contact support team on this, thanks for the answer.

GitPetri, 2023-07-18:
Only now had a chance to do next print on the device. Firmware 1.7.1 and due to sunspots or like I don't see the issue anymore.

GitPetri, 2023-07-18:
.

closed 2023-07-18 by GitPetri