prusa3d/Prusa-Firmware-SL1 #77: Unpredictable number of first layers

cybergurke, 2022-12-13:
I'm trying to get rid of the elephant foot like few others https://github.com/prusa3d/PrusaSlicer/issues/7970.

I have activated the elephant foot compensation in PrusaSlicer, which also works on the first few layers, but I still have a big elephant foot above it. 
According to the description of faded Layers in PrusaSlicer, the printer should fade from the initial exposure time to the normal exposure time within the set number of faded layers. But the printer exposes the first layers completely with the initial exposure time and only then starts to fade. The elephant foot correction fades only to 80% of the layers set on faded layers which results in a lip.

I'm using a SL1s. some people have written that they had no problem with elephant foots on the SL1.

I think this has something to do with a feature that was included with the first sl1s firmware release. 

https://github.com/prusa3d/Prusa-Firmware-SL1/releases/tag/1.6.0
>  Speed of first layers
> 
> While printing the first few layers, it is crucial to ensure that the print adheres perfectly to the printing platform. We have optimized the surface of the platform to a point where it can lift an unsecured resin tank (don't try this at home). Yet, in some scenarios, the first few layers of the object require slightly more time, which is automatically added by the printer. The effect on the overall print time is almost negligible.

Prusa-Support, 2023-01-02:
Hello,
Thank you for your message!
Please write to our support info@prusa3d.com and also attach some pictures of what you see on your prints. The .3mf file would help a lot and would like to start a test print

danbraendle, 2023-02-03:
Hi,
It seems like that this is connected to this thread which I just updated: 
[https://github.com/prusa3d/PrusaSlicer/issues/7970](url)

danbraendle, 2023-12-04:
@cybergurke 👍 Full support from my side, please give this issue more priority. I'm not able to find my issue, it seems like it is deleted?

I just reached out via a customer survey and mentioned this issue here again. Hopefully we come closer to a solution this time.

cybergurke, 2023-12-04:
The link is broken, I forgot to mention it. 
You have to copy the text. There is another link behind the text.  
I hope this one works: https://github.com/prusa3d/PrusaSlicer/issues/7970