prusa3d/Prusa-Firmware-SL1 #78: SL1S running firmware@`1.7.1` returns API version `0.1`

john-rodewald, 2023-01-03:
Querying my Prusa SL1S for `/api/version` yields the following result (truncated):

```
{
    "api": "0.1",
    "hostname": "prusa-sl1s",
    "server": "1.1.0",
    "text": "Prusa SLA 1.0.5"
}
```

This is a problem because there are tools that integrate with Prusa printers which ensure the API version is valid.

Is this expected behaviour? It also seems strange that the firmware version `1.7.1` is nowhere to be found.

The SL1S is several weeks old and has only seen one firmware update, namely from `1.7.0` to `1.7.1`.

Prusa-Support, 2023-01-11:
thank you for your post! Everything looks normal, that's what we expect to see. 
Can you please explain what validation tools you're using? 
Based on your response we might be able to make adjustments

john-rodewald, 2023-01-12:
> thank you for your post! Everything looks normal, that's what we expect to see. Can you please explain what validation tools you're using? Based on your response we might be able to make adjustments

Interesting. Where can I find more information on the topic? I have these questions in particular:

- Why is the API at such a low version? Is the SL1S not a mature product at this point? 
- How does the SL1S v0.1 API differ from the v2 API that runs on other Prusa printers? Are they cross-compatible?
- What's missing to make the jump to v1?